jperkin/node-rpio #112: Livolo light switch code sample

muscaiu, 2020-01-20:
Hello, i found this really simple tutorial on turining on/off a Livolo light switch https://www.instructables.com/id/Controlling-a-Livolo-RF-Light-Switch-Using-a-Raspb/

It's the only code i found working, however the problem is i can't really read python and i was wondering if **someone can translate this into node maybe using this library**.

Thanks!
 
```
import time
import sys
import RPi.GPIO as GPIO

off =  '1242424352424342424242424242425342524342'
on  =  '124242435242434242424242424242425243424242'

if sys.argv[1:] == 'off':
    NUM_ATTEMPTS = 1000
else:
    NUM_ATTEMPTS = 150

TRANSMIT_PIN = 17

def transmit_code(code):
    '''Transmit a chosen code string using the GPIO transmitter'''
    GPIO.setmode(GPIO.BCM)
    GPIO.setup(TRANSMIT_PIN, GPIO.OUT)
    for t in range(NUM_ATTEMPTS):
        for i in code:
            if i == '1':
                GPIO.output(TRANSMIT_PIN, 1)
                time.sleep(.00055);
                GPIO.output(TRANSMIT_PIN, 0)
            elif i == '2':
                GPIO.output(TRANSMIT_PIN, 0)
                time.sleep(.00011);
                GPIO.output(TRANSMIT_PIN, 1)
            elif i == '3':
                GPIO.output(TRANSMIT_PIN, 0)
                time.sleep(.000303);
                GPIO.output(TRANSMIT_PIN, 1)
            elif i == '4':
                GPIO.output(TRANSMIT_PIN, 1)
                time.sleep(.00011);
                GPIO.output(TRANSMIT_PIN, 0)
            elif i == '5':
                GPIO.output(TRANSMIT_PIN, 1)
                time.sleep(.00029);
                GPIO.output(TRANSMIT_PIN, 0)
            else:
                continue
        GPIO.output(TRANSMIT_PIN, 0)
    GPIO.cleanup()

if __name__ == '__main__':
    for argument in sys.argv[1:]:
        exec('transmit_code(' + str(argument) + ')')

# How to use:: I am putting this here, because it seems few people have figured out livolo switches.
# Hold down livolo light switch for 5 seconds and wait for a beep.
# Run ```python livolo.py on```
# Livolo light switch should beep again, signalling it is paired
# Now running ```python livolo.py on``` will toggle the switch.
# ```python livolo.py off``` is global to all switches and should work out of box.

# The following lines are possible RF codes which can be learned by the switch.
# Just make them strings and save them to variables similarly to 'on' or 'off'.
# I only have 1 switch so multiple RF freqs werent necessary.
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 3, 5, 3, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 3, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 3, 4, 2, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 3, 4, 2, 5, 3, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 3, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2
# 1, 2, 4, 2, 4, 2, 4, 3, 5, 2, 4, 2, 4, 3, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 4, 2, 5, 2, 4, 3, 4, 2, 4, 2, 4, 2
```

jperkin, 2020-01-21:
This should be pretty straight-forward:

* Replace `GPIO.output()` with `rpio.write()`
* Replace `time.sleep()` with `rpio.usleep()` and convert to microseconds.

I'd avoid using the BCM numbering, just use the physical numbering that rpio defaults to and set the physical pin you are using instead for `TRANSMIT_PIN` (in the case of BCM17 that would be physical pin 11).

The rest is basic JS.

closed 2020-01-21 by jperkin

muscaiu, 2020-03-22:
When someone sais pretty **straight-forward**, i't's never like that at all! :D
I have no experience with Python, so pretty much all the code is unclear.