jperkin/node-rpio #116: Cannot resolve fs when trying to compile

maximus1127, 2020-05-04:
Hi there, I'm very excited to use your package. It's very well documented and seems very powerful. I'm having a problem though. When I try to compile, it gives me this error: 

```
ERROR in ./node_modules/bindings/bindings.js
Module not found: Error: Can't resolve 'fs' in '/var/www/myproject/node_modules/bindings'
 @ ./node_modules/bindings/bindings.js 5:9-22
 @ ./node_modules/rpio/lib/rpio.js
 @ ./resources/js/app.js
 @ multi ./resources/js/app.js ./resources/sass/app.scss

ERROR in ./node_modules/rpio/lib/rpio.js
Module not found: Error: Can't resolve 'fs' in '/var/www/myproject/node_modules/rpio/lib'
 @ ./node_modules/rpio/lib/rpio.js 18:9-22
 @ ./resources/js/app.js
 @ multi ./resources/js/app.js ./resources/sass/app.scss

```

I don't know how to fix this as the 'fs' module doesn't even seem to be available to download anymore. Can you offer any advice?

jperkin, 2020-05-04:
`fs` is a core node module. Looking around for similar error messages suggests you might be using webpack or similar? This module only works on real hardware, you can't run it in a browser.

maximus1127, 2020-05-04:
oh really? yes you're right, i am using webpack. this is my first time exploring a project like this, could I trouble you for some advice? what do you mean "real hardware"? I'm hosting this project on the pi itself with apache. I'm trying to use WebRTC to have real time controls. As that's based on javascript, I was trying to control the GPIO's through browser based data signals. Is there anyway that you're aware of to do this? Or do I just have to make ajax calls to local python scripts?

neophob, 2020-09-08:
`webpack-node-externals` might be something for you

jperkin, 2020-09-09:
By real hardware I mean that the module requires direct access to the file system, specifically interacting with the kernel via `/dev/mem` or `/dev/gpiomem`, and any browser-based environment will not permit you to do this.

Usually if you want to write a web application you'd split it into two parts, one running on the server talking to the hardware and exposing a REST API, then a frontend that talks to the API over HTTP and provides an interface for the user to interact with it.

closed 2020-09-09 by jperkin