jperkin/node-rpio #120: MCP23017 example request

jas777, 2020-06-23:
Hello, I have a problem with MCP23017 - I've kinda "copied" python example code into JS and it doesn't work. Does anybody have a working example?

jperkin, 2022-04-09:
I don't have any of these, sorry.

closed 2022-04-09 by jperkin

sarink, 2023-07-10:
@jas777 did you ever get it to work?
@jperkin Does this lib support using a port expander (or, is there any reason it wouldn't)?

jperkin, 2023-07-11:
Not directly, it only knows about native pins, but there's no reason why you can't use it to drive one.

jas777, 2023-07-11:
> @jas777 Did you ever get it to work? @jperkin Does this lib support using a port expander (or, is there any reason it wouldn't)?

I ended up using `i2c-bus`, you can check my implementation at https://github.com/jas777/quassendorf/blob/main/src/mcp/MCP23017.ts