jperkin/node-rpio #124: Rpio.i2cReadRegisterRestart() not working

chandic, 2020-07-19:
There is no binding function for **i2cReadRegisterRestart**, **binding.i2c_read_reg_rs** is undefiled in rpio.js 

Error:
bh1750, File: bh1750_module.js, Function: readLight, TypeError: bindfunc is not a function
    at bindcall3 (/home/pi/Desktop/light-sensor/node_modules/rpio/lib/rpio.js:120:9)
    at rpio.i2cReadRegisterRestart (/home/pi/Desktop/light-sensor/node_modules/rpio/lib/rpio.js:715:9)

jperkin, 2020-07-20:
This should be fixed in https://github.com/jperkin/node-rpio/commit/1580f3d30026bfe84a2e135f6bbb7b523ad0193e and in the 2.1.3 release.

closed 2020-07-20 by jperkin

chandic, 2020-07-23:
Still have issues. You have exported the method **i2c_read_register_rs** from rpio.c but forgot to update the method name in rpio.js which is currently **i2c_read_reg_rs**.

jperkin, 2020-07-23:
Hah thanks, I've fixed this in 2.2.0, can you try that?