jperkin/node-rpio #129: Loading non-context-aware native module in renderer (N-API Support)

devrim-oguz, 2020-09-08:
`Uncaught Error: Loading non-context-aware native module in renderer: '/home/alba/Automat_ArGe/node_modules/mfrc522-rpi/node_modules/rpio/build/Release/rpio.node', but app.allowRendererProcessReuse is true. See https://github.com/electron/electron/issues/18397.
    at process.func [as dlopen] (electron/js2c/asar.js:140)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1196)
    at Object.func [as .node] (electron/js2c/asar.js:140)
    at Module.load (internal/modules/cjs/loader.js:981)
    at Module._load (internal/modules/cjs/loader.js:881)
    at Function.Module._load (electron/js2c/asar.js:769)
    at Module.require (internal/modules/cjs/loader.js:1023)
    at require (internal/modules/cjs/helpers.js:77)
    at bindings (/home/alba/Automat_ArGe/node_modules/bindings/bindings.js:112)
    at Object.<anonymous> (/home/alba/Automat_ArGe/node_modules/mfrc522-rpi/node_modules/rpio/lib/rpio.js:17)`

I'm getting the following error while trying to run my application in Electron. This is probably due to a lack of N-API support on the library. Can anybody help me how to fix this? Or is N-API support going to be added?

devrim-oguz, 2020-09-14:
Gives the same error with different libraries as well. Any way to add this support please?

```
/home/alba/Interface_Automat_ArGe/node_modules/bindings/bindings.js:121 Uncaught Error: Loading non-context-aware native module in renderer: '/home/alba/Interface_Automat_ArGe/node_modules/rpio/build/Release/rpio.node', but app.allowRendererProcessReuse is true. See https://github.com/electron/electron/issues/18397.
    at process.func [as dlopen] (electron/js2c/asar.js:140)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1196)
    at Object.func [as .node] (electron/js2c/asar.js:140)
    at Module.load (internal/modules/cjs/loader.js:981)
    at Module._load (internal/modules/cjs/loader.js:881)
    at Function.Module._load (electron/js2c/asar.js:769)
    at Module.require (internal/modules/cjs/loader.js:1023)
    at require (internal/modules/cjs/helpers.js:77)
    at bindings (/home/alba/Interface_Automat_ArGe/node_modules/bindings/bindings.js:112)
    at Object.<anonymous> (/home/alba/Interface_Automat_ArGe/node_modules/rpio/lib/rpio.js:17)
```

Thanks in advance

jperkin, 2020-09-14:
Probably not, no.  Running rpio in an Electron context doesn't make much sense.

devrim-oguz, 2020-09-14:
What do you mean not?? Some people might want to use gpio with graphical applications. And the npm library mfrc522-rpi depends on this repository. Which is used for RFID cards. So graphical applications eith RFID or 
GPIO functionality are a thing. I can help to contribute to this code if you point me to some directions as well.

devrim-oguz, 2020-09-14:
What is it usefull to access raspberry GPIO only with nodejs if you aren't going to use any interface?

jperkin, 2020-09-14:
To use rpio you need full access to the host file system and character devices, and it's likely the Electron environment will not allow you to do that.  So you need to split your application into a server component that talks to the hardware, and then a client component that provides a graphical interface for the user.

devrim-oguz, 2020-09-14:
It doesn't bother electron, only nodejs. In fact I have a running application right now but I have to use `app.allowRendererProcessReuse = false` which is deprecated and will be removed in later versions. I just change file permissions under linux to let user access gpio functionalities and it works. It just so that this library needs to support context awareness for it to load properly.

jperkin, 2020-09-15:
From a very brief look it seems that N-API is an alternative to NaN, but only supports the latest LTS nodejs releases.  I have zero interest in doing that, one of the core principles of this module is to be as portable as possible, so I think if you want to do this you're going to have to fork your own project and completely rewrite it to use N-API.

devrim-oguz, 2020-09-15:
Alright, thank you. I might fork it in the future. I'm closing the issue then.

closed 2020-09-15 by devrim-oguz