jperkin/node-rpio #130: Provide Prebuild binaries

neophob, 2020-09-09:
I'm working on a Electron app, mainly on OSX but want to run on a RPi (obviously). Anyway by RPi build from my OSX did not work as expected, after some debugging I ended up on https://www.electron.build/multi-platform-build:

If your app has native dependency, it can be compiled only on the target platform unless prebuild is not used.
prebuild is a solution, but most node modules don’t provide prebuilt binaries.

Tbh I'm not really sure what this exactly means - IF prebuild is used "crosscompile" works?

jperkin, 2020-09-09:
This won't work.  You can use rpio in mock mode to test on macOS, but proper use will require direct access to the hardware on Linux.

closed 2020-09-09 by jperkin

neophob, 2020-09-09:
thanks @jperkin for your feedback. I like the mock feature very much on OSX. 

The intend of this issue is to provide compiled nodejs binaries for the rpi (armv7l) - with prebuild (https://www.npmjs.com/package/prebuild). This would make installing the library easier and it would "just work" for electron as example. What's you view about that?

jperkin, 2020-09-09:
I don't like it, but there's nothing stopping you from building your own.