jperkin/node-rpio #132: Pin control on RPi Compute module does not work

Papouchcom, 2020-10-06:
Pin control on the Raspberry Compute module 3+ does not work. This code:
```
    rpio.init({
        gpiomem: true,
        mapping: 'physical',
        mock: undefined,
        close_on_exit: true
    });
    rpio.open(70, rpio.OUTPUT, rpio.HIGH);
```
...crashes with this error:
**Pin 70 is not valid when using physical mapping**
(GPIO mode with pin number setting of 40 does not work either.)

jperkin, 2020-10-06:
Yeh, I've never had any users with a Compute Module who can verify what the pinmap is, or what it detects as.

Could you fill in the values from https://gist.github.com/jperkin/c37a574379ef71e339361954be96be12 to help out, and ideally verify the correct pin mapping?  It should be pretty easy to then add support.  Thanks.

Papouchcom, 2020-10-06:
Thanks a lot for the quick response! I added logs from different versions of CM to the comments in [gist](https://gist.github.com/jperkin/c37a574379ef71e339361954be96be12#gistcomment-3479014).

I think the pin map for CM is this (on page 9):
https://www.raspberrypi.org/documentation/hardware/computemodule/datasheets/rpi_DATA_CM3plus_1p0.pdf#page=10
How could I try it, please?

Papouchcom, 2020-10-09:
Here is a modified file with a array (PINMAP_200) of GPIO pins on the CM:
[rpio.zip](https://github.com/jperkin/node-rpio/files/5353128/rpio.zip)

jperkin, 2020-10-09:
Thanks!  Have you verified it works correctly?  If so I'll merge it.

Papouchcom, 2020-10-09:
I tested only with CM 3+ and only with two GPIO pins. I also modified the module detection, but I don't know if it detects standard RPi correctly.

jperkin, 2021-01-18:
Support was added a while back, and should now include CM4 too.

closed 2021-01-18 by jperkin

shaynem, 2022-01-07:
I've been trying to figure out the problem I was having when using the sourcekit PiTray - it exposes the default 40 pin GPIO, therefore I had to change

 line 577

```PINMAP_40```

I'm not sure if there is a way to detect the PITray pragmatically, however if I do a PR, would you accept config done via 

```rpio.init([options])```

```pinmap: 'PINMAP_40'```

or another suggestion?