jperkin/node-rpio #134: Raspberry Pi Compute Module 4 Rev 1.0

JesperNaarttijarvi, 2021-01-14:
Does it work for the CM4? I get an error when trying to adapt our CM3 implementations to the CM4. I can't read the PN532 module via node anymore.

Board: Raspberry Pi Compute Module 4 Rev 1.0
Hardwar: BCM2711

jperkin, 2021-01-14:
The code should match and enable the 200-pin pinmap, but I don't know if there were any changes since the CM3.

Unfortunately I don't have any CM boards to test on.

JesperNaarttijarvi, 2021-01-14:
I'll try to dive deeper into it and see if I find exactly where the problem lies, how ever might not be skilled enough to find it.

jperkin, 2021-01-14:
The pinmap is here: https://github.com/jperkin/node-rpio/blob/master/lib/rpio.js#L232-L334

It'll be a case of finding the GPIO documentation for the BCM2711 and then validating that the map between physical pin and gpio pin is correct, then if wrong try plumbing in the correct values and see if it works.

JesperNaarttijarvi, 2021-01-14:
That was the indeed the problem. For anyone interested in a quick fix, here is the pinout i used for cm4.

`/*
	 * Compute Module 4.
	 */
	PINMAP_200: [
		-1,
		-1, -1,		/*  P1  P2 */
		-1, -1,		/*  P3  P4 */
		-1, -1,		/*  P5  P6 */
		-1, -1,		/*  P7  P8 */
		-1, -1,		/*  P9  P10 */
		-1, -1,		/* P11  P12 */
		-1, -1,		/* P13  P14 */
		-1, -1,		/* P15  P16 */
		-1, -1,		/* P17  P18 */
		-1, -1,		/* P19  P20 */
		-1, -1,		/* P21  P22 */
		-1, 26,		/* P23  P24 */
		21, 19,		/* P25  P26 */
		20, 13,		/* P27  P28 */
		16, 6,		/* P29  P30 */
		12, -1,		/* P31  P32 */
		-1, 5,		/* P33  P34 */
		1, 0,		/* P35  P36 */
		7, 11,		/* P37  P38 */
		8, 9,		/* P39  P40 */
		25, -1,		/* P41  P42 */
		-4, 10,		/* P43  P44 */
		24, 22,		/* P45  P46 */
		23, 27,		/* P47  P48 */
		18, 17,		/* P49  P50 */
		15, -1,		/* P51  P52 */
		-1, 4,		/* P53  P54 */
		14, 3,		/* P55  P56 */
		-1, 2,		/* P57  P58 */
		-1, -1,		/* P59  P60 */
		-1, -1,		/* P61  P62 */
		-1, -1,		/* P63  P64 */
		-1, -1,		/* P65  P66 */
		-1, -1,		/* P67  P68 */
		-1, -1,		/* P69  P70 */
		-1, -1,		/* P71  P72 */
		-1, -1,		/* P73  P74 */
		-1, -1,		/* P75  P76 */
		-1, -1,		/* P77  P78 */
		-1, 45,		/* P79  P80 */
		-1, 44,		/* P81  P82 */
		-1, -1,		/* P83  P84 */
		-1, -1,		/* P85  P86 */
		-1, -1,		/* P87  P88 */
		-1, -1,		/* P89  P90 */
		-1, -1,		/* P91  P92 */
		-1, -1,		/* P93  P94 */
		-1, -1,		/* P95  P96 */
		-1, -1,		/* P97  P98 */
		-1, -1,		/* P99  P100 */
		-1, -1,		/* P101 P102 */
		-1, -1,		/* P103 P104 */
		-1, -1,		/* P105 P106 */
		-1, -1,		/* P107 P108 */
		-1, -1,		/* P109 P110 */
		-1, -1,		/* P111 P112 */
		-1, -1,		/* P113 P114 */
		-1, -1,		/* P115 P116 */
		-1, -1,		/* P117 P118 */
		-1, -1,		/* P119 P120 */
		-1, -1,		/* P121 P122 */
		-1, -1,		/* P123 P124 */
		-1, -1,		/* P125 P126 */
		-1, -1,		/* P127 P128 */
		-1, -1,		/* P129 P130 */
		-1, -1,		/* P131 P132 */
		-1, -1,		/* P133 P134 */
		-1, -1,		/* P135 P136 */
		-1, -1,		/* P137 P138 */
		-1, -1,		/* P139 P140 */
		-1, -1,		/* P141 P142 */
		-1, -1,		/* P143 P144 */
		-1, -1,		/* P145 P146 */
		-1, -1,		/* P147 P148 */
		-1, -1,		/* P149 P150 */
		-1, -1,		/* P151 P152 */
		-1, -1,		/* P153 P154 */
		-1, -1,		/* P155 P156 */
		-1, -1,		/* P157 P158 */
		-1, -1,		/* P159 P160 */
		-1, -1,		/* P161 P162 */
		-1, -1,		/* P163 P164 */
		-1, -1,		/* P165 P166 */
		-1, -1,		/* P167 P168 */
		-1, -1,		/* P169 P170 */
		-1, -1,		/* P171 P172 */
		-1, -1,		/* P173 P174 */
		-1, -1,		/* P175 P176 */
		-1, -1,		/* P177 P178 */
		-1, -1,		/* P179 P180 */
		-1, -1,		/* P181 P182 */
		-1, -1,		/* P183 P184 */
		-1, -1,		/* P185 P186 */
		-1, -1,		/* P187 P188 */
		-1, -1,		/* P189 P190 */
		-1, -1,		/* P191 P192 */
		-1, -1,		/* P193 P194 */
		-1, -1,		/* P195 P196 */
		-1, -1,		/* P197 P198 */
		-1, -1		/* P199 P200 */
	],`

closed 2021-01-14 by JesperNaarttijarvi

jperkin, 2021-01-14:
Thanks, do you have a reference for this?

JesperNaarttijarvi, 2021-01-15:
I found the pin out in the [official datasheet](https://datasheets.raspberrypi.org/cm4/cm4-datasheet.pdf) and mapped the GPIO ports.

jperkin, 2021-01-15:
Ah perfect, thanks, I'll add proper support for this shortly.

jperkin, 2021-01-15:
Released in version 2.4.0, many thanks!

JesperNaarttijarvi, 2021-01-15:
Nice, will try to update next week and see if it works.