jperkin/node-rpio #136: Last version - Invalid arg2 on read

FerX, 2021-01-17:
Raspberry PI Zero

with update to the latest version, when I send the read command I get this error.
before, it worked fine

```
       return bindfunc(arg1, arg2);
               ^

TypeError: Invalid arg2
    at bindcall2 (/home/meteo/meteostation/node_modules/rpio/lib/rpio.js:118:9)
    at rpio.read (/home/meteo/meteostation/node_modules/rpio/lib/rpio.js:891:9)
    at pollVento (/home/meteo/meteostation/test.js:46:12)
    at rpio.pincb (/home/meteo/meteostation/node_modules/rpio/lib/rpio.js:963:4)
    at rpio.emit (events.js:198:13)
    at Timeout.event_poll [as _onTimeout] (/home/meteo/meteostation/node_modules/rpio/lib/rpio.js:529:19)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

JohnRucker, 2021-01-17:
Im getting the same thing.  Huge, Huge problem for me!!

JohnRucker, 2021-01-17:
Rolled back to version 2.2.0 and working again.  Wonder what is going on with this library??  It has been rock solid for years!  Just an FYI if anyone cares.  Version 2.4.0 caused my Pi Zero to pretty much brick itself.  Thank god I caught it before it got pushed out to the field.  I hope others are lucky too.

jperkin, 2021-01-17:
Oops, I'm really sorry. I've been running the latest code on a number of devices, but none of them use the `read()` method so I didn't notice.

This should fix the problem, can you verify?

```diff
diff --git a/lib/rpio.js b/lib/rpio.js
index b226491..49b650f 100755
--- a/lib/rpio.js
+++ b/lib/rpio.js
@@ -889,7 +889,7 @@ rpio.prototype.read = function(pin, mode)
        if (mode === undefined)
                mode = false;
 
-       return bindcall2(binding.gpio_read, pin_to_gpio(pin), mode);
+       return bindcall2(binding.gpio_read, pin_to_gpio(pin), mode | 0);
 }
 
 rpio.prototype.readbuf = function(pin, buf, len, mode)
```

jperkin, 2021-01-18:
I've pushed version 2.4.1 to fix this, sorry again for the breakage.

closed 2021-01-18 by jperkin

JohnRucker, 2021-01-18:
I updated to 2.4.1 and pushed it to my test machine.  Getting a new error now looks like in line 892 of rpio.js.  Rolled back to 2.2.0 and everything is working again.  Thanks for getting on this so quick sorry I didn't test it sooner. 

```
/opt/rGauge/gdtMan/node_modules/rpio/lib/rpio.js:892
	return bindcall2(binding.gpio_read, pin_to_gpio(pin), Number(mode);
	                                                                 ^

SyntaxError: missing ) after argument list
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/rGauge/gdtMan/lib/LEDClass.js:1:21)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)

```

jperkin, 2021-01-18:
Gah, stupid mistake copying the change from my test host.  Sorry, this is really poor of me, fixing now.

jperkin, 2021-01-18:
I just published version 2.4.2 which should fix this.  Really sorry again, I need to sit down and finish off the migration from Travis CI so that I get automated tests back up and running again to catch stuff like this (as well as not rushing to fix breakages and introducing more breakages...)

JohnRucker, 2021-01-18:
2.4.2 Looks great!  Thanks again for getting on this so quick.  Don't be so hard on yourself!  You do fantastic work my friend!!