jperkin/node-rpio #150: Could not locate the bindings file

markwbrown, 2021-08-13:
Ubuntu 20.04 on Raspberry Pi 3B with Node v14.17.5, I get the following:  

Error: Could not locate the bindings file. Tried:
  → /path/to/dir/node_modules/rpio/build/rpio.node
  → /path/to/dir/node_modules/rpio/build/Debug/rpio.node
  → /path/to/dir/node_modules/rpio/build/Release/rpio.node
  → /path/to/dir/node_modules/rpio/out/Debug/rpio.node
  → /path/to/dir/node_modules/rpio/Debug/rpio.node
  → /path/to/dir/node_modules/rpio/out/Release/rpio.node
  → /path/to/dir/node_modules/rpio/Release/rpio.node
  → /path/to/dir/node_modules/rpio/build/default/rpio.node
  → /path/to/dir/node_modules/rpio/compiled/14.17.5/linux/arm64/rpio.node
  → /path/to/dir/node_modules/rpio/addon-build/release/install-root/rpio.node
  → /path/to/dir/node_modules/rpio/addon-build/debug/install-root/rpio.node
  → /path/to/dir/node_modules/rpio/addon-build/default/install-root/rpio.node
  → /path/to/dir/node_modules/rpio/lib/binding/node-v83-linux-arm64/rpio.node
     at bindings (/path/to/dir/node_modules/bindings/bindings.js:126:9)
     at Object.<anonymous> (/path/to/dir/node_modules/rpio/lib/rpio.js:17:34)
     at Module._compile (internal/modules/cjs/loader.js:1072:14)
     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
     at Module.load (internal/modules/cjs/loader.js:937:32)
     at Function.Module._load (internal/modules/cjs/loader.js:778:12)
     at Module.require (internal/modules/cjs/loader.js:961:19)
     at Module.Hook._require.Module.require (/home/git/.nvm/versions/node/v14.17.5/lib/node_modules/pm2/node_modules/require-in-the-middle/index.js:80:39)
     at require (internal/modules/cjs/helpers.js:92:18)
     at Object.<anonymous> (/path/to/dir/index.js:4:14) {
   tries: [
     '/path/to/dir/node_modules/rpio/build/rpio.node',
     '/path/to/dir/node_modules/rpio/build/Debug/rpio.node',
     '/path/to/dir/node_modules/rpio/build/Release/rpio.node',
     '/path/to/dir/node_modules/rpio/out/Debug/rpio.node',
     '/path/to/dir/node_modules/rpio/Debug/rpio.node',
     '/path/to/dir/node_modules/rpio/out/Release/rpio.node',
     '/path/to/dir/node_modules/rpio/Release/rpio.node',
     '/path/to/dir/node_modules/rpio/build/default/rpio.node',
     '/path/to/dir/node_modules/rpio/compiled/14.17.5/linux/arm64/rpio.node',
     '/path/to/dir/node_modules/rpio/addon-build/release/install-root/rpio.node',
     '/path/to/dir/node_modules/rpio/addon-build/debug/install-root/rpio.node',
     '/path/to/dir/node_modules/rpio/addon-build/default/install-root/rpio.node',
     '/path/to/dir/node_modules/rpio/lib/binding/node-v83-linux-arm64/rpio.node'
   ]
 }

markwbrown, 2021-08-13:
Installing globally via `npm install -g rpio` does not work. Navigating to the rpio folder within node_modules and issuing `npm install` creates a build folder.

jperkin, 2022-04-09:
Yeh I wouldn't ever recommend performing global installs, there are many issues with doing so.  Just create a project area, `npm install rpio`, and then run from there.

closed 2022-04-09 by jperkin