jperkin/node-rpio #151: Cannot use PWM

Girum, 2021-08-24:
Hi!

I follow the tutorial, and use this code

`var rpio = require('rpio')

var express = require('express')

var app = express()

const pinsPWM = [12]

rpio.init({gpiomem: false, mapping: 'physical'})

//rpio.close(pinsPWM[0])
//rpio.open(pinsPWM[0], rpio.OUTPUT, rpio.LOW)

rpio.open(pinsPWM[0], rpio.PWM)

rpio.pwmSetClockDivider(64)

rpio.pwmSetRange(pinsPWM[0], 1024)

app.post('/turn_on', (req, res) => {
    //rpio.write(pinsPWM[0], rpio.HIGH)
    rpio.pwmSetData(pinsPWM[0], 1024)
    res.send({msg: "LED turned on!"})
})

app.post('/turn_off', (req, res) => {
    //rpio.write(pinsPWM[0], rpio.LOW)
    rpio.pwmSetData(pinsPWM[0], 0)
    res.send({msg: "LED turned off!"})
})

app.listen(3044, () => {
    console.log("Server has started!")
})`

But the PWM dont work. I even chmod 777 the dev/mem and dev/gpiomem

When I use the rpio.write, it works perfectly. I fresh install the Rpi yesterday. How can I procceed? I used 2 differents pins and get nothing.

npreiser, 2021-12-24:
I am getting the same result, pwm from all pwm pins is not working,  using Node12/  Buster release...

npreiser, 2021-12-24:
Did you ever resolve the issue?,  I am using a pi3B+ , ,  
I have confirmed the pins working fine when just doing ditial toggling,  but pwm mode is not working at all.

filemon, 2022-01-03:
Facing the same problem. rpio npm works great in Balena OS, but doesn't generate any signal in Raspbian. Same node version (10.20) used. Must be something fishy in native libs.

filemon, 2022-01-04:
@npreiser found a solution. PWM started to generate pulses after disabling 'dtparam=audio=on' in /boot/config.txt.