jperkin/node-rpio #164: M1 Mac support?

sarink, 2022-03-13:
```
Error: dlopen(node_modules/rpio/build/Release/rpio.node, 0x0001): tried: 'node_modules/rpio/build/Release/rpio.node' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/usr/local/lib/rpio.node' (no such file), '/usr/lib/rpio.node' (no such file)
```
Will this lib ever support arm architecture?

waltsatan, 2022-03-19:
How would you use this library on a Mac that doesn't have gpio pins?

jperkin, 2022-04-09:
This library only supports Raspberry Pi and similar devices that expose GPIO interfaces.  If there's something in the documentation that lead you to believe it might somehow be useful on Macs or other desktop computers then please let me know.

closed 2022-04-09 by jperkin

sarink, 2022-05-13:
Sorry for the late follow up here. I'm referring to running in mock mode.

> This module will only interface with hardware on Linux, but should compile on any other platform, where it will run in mock mode by default.

On an intel mac, mock mode works fine. On an m1, you receive the above error.

jperkin, 2022-05-13:
In that case I'd check with whatever you are using for install of nodejs, seems like it's defaulting to building `x86_64` objects which is wrong.

If I use the nodejs and npm packages from my pkgsrc repository (https://pkgsrc.joyent.com/install-on-osx/) it works fine:

```
$ uname -rsm
Darwin 21.4.0 arm64

$ npm install rpio
added 4 packages, and audited 5 packages in 12s
found 0 vulnerabilities

$ file node_modules/rpio/build/Release/rpio.node
node_modules/rpio/build/Release/rpio.node: Mach-O 64-bit bundle arm64

$ node -e "var rpio = require('rpio'); rpio.init();"
WARNING: Hardware auto-detect failed, running in raspi-3 mock mode
```