jperkin/node-rpio #166: Orange Pi Zero 2?

Midoxx22, 2022-04-06:
Hi, 
thanks for your awesome work!
Libary works flawless on RPI Zero.

Since it is nearly impossible to get RPI´s zero lately do you plan to implement Orange Pi Zero 2 Support?

Thanks!

jperkin, 2022-04-09:
It should just be a case of adding the correct pinmap, if someone is able to do that and verify it works then I'll merge it in.

samlovebutter, 2023-06-30:
Hello, I have a correct pinmap how do i prove it working?

Here:
 [
		 -1,
		 -1,   -1,	/*  P1  P2 */
		229,   -1,	/*  P3  P4 */
		228,   -1,	/*  P5  P6 */
		 73,  226,	/*  P7  P8 */
		 -1,  227,	/*  P9  P10 */
		 70,   75,	/* P11  P12 */
		 69,   -1,	/* P13  P14 */
		 72,   79,	/* P15  P16 */
		 -1,   78,	/* P17  P18 */
		231,   -1,	/* P19  P20 */
		232,   71,	/* P21  P22 */
		230,  233,	/* P23  P24 */
		 -1,   74,	/* P25  P26 */
	]