jperkin/node-rpio #167: Bindings.js Error

plezhh, 2022-07-24:
Hello,

I have an application with two folders, backend and src, with individuals node_modules folder.

If I require the rpio library in backend all works good.

If I try to require it in src, with npm run build && npm start, I get a blank white page, with the following errors in console:

[image]

Could you help me with some ideas?

Thanks!

plhyhc, 2022-07-25:
The rpio library is designed to be used on the backend only.  For use with the GPIO pins.  The frontend which loads into a web browser does not have access to a the GPIO pins.  If you want to connect the frontend to the backend, the best way to do this is by an Ajax requests to a backend method that handles the GPIO I/O.

plezhh, 2022-07-25:
Ohhh, so that's the problem. Thank you very much!

closed 2022-07-25 by plezhh