jperkin/node-rpio #177: Node 20 Support

michaelgambold, 2023-11-11:
I've been using this library with node 14, 16, 18 however when I have started the process of lifting my project to node 20 (the now LTS) I get the following errors when trying to install rpio.

## Error message
```
npm ERR! code 1
npm ERR! path /Users/michael/Source/node-automation-hat/node_modules/rpio
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/rpio/src/rpio.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.4.0
npm ERR! gyp info using node@20.9.0 | darwin | arm64
npm ERR! gyp info find Python using Python version 3.11.6 found at "/opt/homebrew/opt/python@3.11/bin/python3.11"
npm ERR! gyp info spawn /opt/homebrew/opt/python@3.11/bin/python3.11
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/Users/michael/.nvm/versions/node/v20.9.0/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/michael/Source/node-automation-hat/node_modules/rpio/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/michael/.nvm/versions/node/v20.9.0/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/michael/Library/Caches/node-gyp/20.9.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/michael/Library/Caches/node-gyp/20.9.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/Users/michael/.nvm/versions/node/v20.9.0/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/michael/Library/Caches/node-gyp/20.9.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/michael/Source/node-automation-hat/node_modules/rpio',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/rpio.cc:17:
npm ERR! In file included from ../../nan/nan.h:178:
npm ERR! ../../nan/nan_callbacks.h:55:23: error: no member named 'AccessorSignature' in namespace 'v8'
npm ERR! typedef v8::Local<v8::AccessorSignature> Sig;
npm ERR!                   ~~~~^
npm ERR! In file included from ../src/rpio.cc:17:
npm ERR! ../../nan/nan.h:684:39: warning: 'IdleNotificationDeadline' is deprecated: Use MemoryPressureNotification() to influence the GC schedule. [-Wdeprecated-declarations]
npm ERR!     return v8::Isolate::GetCurrent()->IdleNotificationDeadline(
npm ERR!                                       ^
npm ERR! /Users/michael/Library/Caches/node-gyp/20.9.0/include/node/v8-isolate.h:1291:3: note: 'IdleNotificationDeadline' has been explicitly marked deprecated here
npm ERR!   V8_DEPRECATE_SOON(
npm ERR!   ^
npm ERR! /Users/michael/Library/Caches/node-gyp/20.9.0/include/node/v8config.h:550:39: note: expanded from macro 'V8_DEPRECATE_SOON'
npm ERR! # define V8_DEPRECATE_SOON(message) [[deprecated(message)]]
npm ERR!                                       ^
npm ERR! In file included from ../src/rpio.cc:17:
npm ERR! ../../nan/nan.h:2544:8: error: no matching member function for call to 'SetAccessor'
npm ERR!   tpl->SetAccessor(
npm ERR!   ~~~~~^~~~~~~~~~~
npm ERR! /Users/michael/Library/Caches/node-gyp/20.9.0/include/node/v8-template.h:816:8: note: candidate function not viable: no known conversion from 'imp::Sig' (aka 'int') to 'SideEffectType' for 7th argument
npm ERR!   void SetAccessor(
npm ERR!        ^
npm ERR! /Users/michael/Library/Caches/node-gyp/20.9.0/include/node/v8-template.h:809:8: note: candidate function not viable: no known conversion from 'imp::NativeGetter' (aka 'void (*)(v8::Local<v8::Name>, const v8::PropertyCallbackInfo<v8::Value> &)') to 'AccessorGetterCallback' (aka 'void (*)(Local<String>, const PropertyCallbackInfo<Value> &)') for 2nd argument
npm ERR!   void SetAccessor(
npm ERR!        ^
npm ERR! 1 warning and 2 errors generated.
npm ERR! make: *** [Release/obj.target/rpio/src/rpio.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/Users/michael/.nvm/versions/node/v20.9.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:203:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:514:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:294:12)
npm ERR! gyp ERR! System Darwin 23.0.0
npm ERR! gyp ERR! command "/Users/michael/.nvm/versions/node/v20.9.0/bin/node" "/Users/michael/.nvm/versions/node/v20.9.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/michael/Source/node-automation-hat/node_modules/rpio
npm ERR! gyp ERR! node -v v20.9.0
npm ERR! gyp ERR! node-gyp -v v9.4.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in: /Users/michael/.npm/_logs/2023-11-11T21_03_12_160Z-debug-0.log
```

## Setup

Although this error is from my M1 macbook (with NVM) this also fails in GitHub Action CI.

hoang-danny05, 2024-01-03:
RPIO can only be downloaded on the raspberry pi. If you wish to develop with your mac, you may npm install @types/rpio.

michaelgambold, 2024-01-07:
@hoang-danny05 I tried this to see if it worked but didn't.

From the trace the issue is internal to Node 20 and how rpio interfaces with it (not something as simple as types).

Some other projects have reported this (see below links for discussions regarding this).

https://github.com/nodejs/nan/issues/942
https://github.com/nodejs/nan/pull/943

michaelgambold, 2024-01-07:
Found this from an electron issue about the same. The underlying fix is released in nan version 2.17.0; However rpio is using 2.14.1 (https://github.com/jperkin/node-rpio/blob/52ae53875c3452622a0e3b5130c52f7b85a2099b/package.json#L15)

https://github.com/electron/electron/issues/35193#issuecomment-1273796411

[image]

The underlying fix is just to update the nan dependency. If I have time I might try and get a PR for this.

pdillon, 2024-01-08:
Deleting my lock file with a fresh node_modules install resolved rpio's nan version to 2.18.0

michaelgambold, 2024-01-14:
@pdillon Thanks for that I can confirm that [this comment](https://github.com/jperkin/node-rpio/issues/177#issuecomment-1880268063) worked as a work around.

I still think the minimum version of the library should be updated and I've created PR #180 that includes this change. @jperkin if you would like to review this and push out a new version that would be great.

@jperkin this issue can be closed now regardless of if you choose to do anything with my PR. Just want to make sure you see this issues status before it goes from "open" issues.

DavidEdell, 2025-05-13:
This is an issue for all node versions 20+.

I had to revert back to Node 18 (from 24 on a fresh install) in order to install my modules (of which this is a nested dependency) while trying to migrate a system to a more recent OS install on our Pi.