jperkin/node-rpio #179: Support for Raspberry Pi 5

hoang-danny05, 2024-01-03:
As the new raspberry pi 5 came out, it has come with major hardware differences that make this library incompatible.

absolutetsonga, 2025-02-05:
Hello, I am a newbie working with Raspberry PI and hardware in general. Do you mean that @remarkablearts/rpio is unsupportable with Raspberry Pi 5? If so, what other libraries would you recommend connecting Raspberry hardware with Node.JS/Express backend? Thanks in advance.

hoang-danny05, 2025-02-14:
Hello! To address your first question, yes, the PI 5 will be incompatible with this library because they completely switched the CPU that controls everything. I do not have experience with other Node libraries, and this comment was part of a past project. However, I found a thread that might be useful for you: [RASPBERRY PI FORUM](https://forums.raspberrypi.com/viewtopic.php?t=364182)

I saw that @iiot2k/gpiox claims to provide PI 5 support, but I have not tested it myself! Also, if you find any other solution, please tell me!

gdorbes, 2025-10-31:
See https://www.npmjs.com/package/rpi-io a new module that supports recent versions of Raspberry and related OS and middleware.